fougue/mayo #308: cannot detect opencascade version

FlynnHHH, 2024-11-27:
I am compiling mayo develop branch against opencascade7.8.0 on X64 Windows and cmake failed to detect opencascade's version.

Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.22631.
GIT_EXECUTABLE=D:/Git/bin/git.exe
Mayo_PostBuildCopyRuntimeDLLs_DefaultValue=ON
Qt version 5.14.2
OpenCascade version ..
glTF reader disabled because OpenCascade < v7.4
OBJ reader disabled because OpenCascade < v7.4
glTF writer disabled because OpenCascade < v7.5
OBJ writer disabled because OpenCascade < v7.6
VRML reader disabled because OpenCascade < v7.7
Configuring done (0.5s)

And when building the generated .sln, VS2022 met with error:
LINK : fatal error LNK1104: cannot open file“D:\library\OCCT\OCCT-7_8_0\build\win64\vc14\lib\TKIGES.lib”
However, the TKIGES.lib has been renamed to TKDEIGES.lib in opencascade7.8.0, so probably the cmakelists.txt is to be modified.

HuguesDelorme, 2024-11-27:
Mayo builds fine with OpenCascade 7.8 on Windows, see the current Windows CI status:
https://github.com/fougue/mayo/actions/runs/11795444647

Mayo cmake script just doesn't correctly identify your OpenCascade build
You built OpenCascade yourself, make sure you followed the whole procedure, especially the "install" step(consider the `INSTALL_DIR` variable of OpenCascade cmake script)
https://dev.opencascade.org/doc/overview/html/build_upgrade__building_occt.html

If it still doesn't work then just install and use prebuilt OpenCascade binaries:
https://dev.opencascade.org/release/open-cascade-technology-780

FlynnHHH, 2024-11-30:
Thanks for your reply. Now I just manually set the OpenCascade version to 7.8.0 as a work around. There is something wrong with my CMake configuration when building mayo I guess.

closed 2024-11-30 by FlynnHHH