fougue/mayo #313: GLB to OBJ / MTL issue

fabiencomte, 2025-01-04:
Hello,
I try to convert GLB files with textures in OBJ / MTL and the application goes in infinite loop. No file is created.
[2019_bugatti_divo.zip](https://github.com/user-attachments/files/18306485/2019_bugatti_divo.zip)

[2014__volkswagen_gti_roadster_vision_gt_gr.3.glb.zip](https://github.com/user-attachments/files/18306484/2014__volkswagen_gti_roadster_vision_gt_gr.3.glb.zip)

fabiencomte, 2025-01-04:
Not better with GLTF source.
[2021_bugatti_chiron_pur_sport.zip](https://github.com/user-attachments/files/18306499/2021_bugatti_chiron_pur_sport.zip)

stellarpower, 2025-01-29:
I am reproducing the same behaviour, it locks up. I have had success with some files before. IDK enough about the file types to know what the differences would be, but GLB exported form Blender 3.4 worked fine, GLB export from Blender 4.3 causes the loop. I have also tried converting the GLB to GLTF and the problem the same. I'll see if I can look into it

stellarpower, 2025-01-29:
Okay, so [this](https://github.com/Open-Cascade-SAS/OCCT/blob/V7_8_1/src/TopoDS/TopoDS_Shape.hxx#L94) is the cause of the error, and it's hanging because it's not catching the exception. The worker thread exits presumably with the unhandled exception, but the QApplication thread is hanging waiting for a progress update or something like it.

The following update gets us a clean exit:

```
bool OccObjWriter::writeFile(const FilePath& filepath, TaskProgress* progress) try{

    ... 

    if (m_seqRootLabel.IsEmpty())
        return writer.Perform(m_document, fileInfo, occProgress->Start());
    else
        return writer.Perform(m_document, m_seqRootLabel, nullptr, fileInfo, occProgress->Start());

} catch (const Standard_Failure &e){
    cerr << e.GetMessageString() << "\n\n" << e.GetStackString() << endl;
    return false;
} catch (...) {
    cerr << "Unhandled exception in conversion" << endl;
    return false;
}
```

I'm just following the backtrace here, @HuguesDelorme you'd know far better than I what it really amounts to re the input file. Here's a mesh to reproduce, in binary format but with an extension to fool GH's CDN.

[Blender 4.3 NS S3R1.glb.txt](https://github.com/user-attachments/files/18593850/Rail1.glb.txt)

I'm reproducing on both OpenCascade 7.8.1 and 7.6.3.

Cheers

fabiencomte, 2025-01-30:
Hello, thank you so much to investigate the issue. Best regards.

HuguesDelorme, 2025-01-30:
> I'm just following the backtrace here, [@HuguesDelorme](https://github.com/HuguesDelorme) you'd know far better than I what it really amounts to re the input file. Here's a mesh to reproduce, in binary format but with an extension to fool GH's CDN.
> Cheers

@stellarpower 
Many thanks for your investigation and the hard work
Will propose a fix soon

HuguesDelorme, 2025-02-24:
Hello @fabiencomte and @stellarpower 
This bug has just been fixed in `develop` branch(see [74f4079](https://github.com/fougue/mayo/commit/74f4079d2cfd28342b5d20fe33a28b9a02332527))
On mesh import with OpenCascade reader, the assembly instance nodes are traversed and those having scale transformations are fixed(forced to 1)
This allows the OpenCascade OBJ exporter to successfully write the output mesh

closed 2025-02-24 by HuguesDelorme

stellarpower, 2025-02-25:
Thanks a lot!! Built and tested.

stellarpower, 2025-02-25:
BTW, am yet to attempt putting multiple binaries into one AppImage, will do soon. It'd probably be nice to et both the CLI and the GUI without needing the space for all the dependencies twice. But at the least, assuming you run it twice, you can probably re-use the same icon in whatever script you run by adjusting the desktop file accordingly, just FYI ;)