fougue/mayo #314: Upload CI builds as artifacts to Github actions

wormyrocks, 2025-01-08:
I see that there's a CI task to make macOS builds. Since there are no official macOS builds anywhere, it would be nice if the output of that task were uploaded as an artifact to the github action: https://github.com/actions/upload-artifact

In particular: I had no problem building mayo on my M1 Mac, but I wanted to send a copy to a friend with an Intel Mac, and ran into issues cross-compiling. Would be much easier to just grab from CI.

HuguesDelorme, 2025-01-20:
Hello @wormyrocks
Thanks for your message and sorry for the late answer
It's been a while since I would like to provide arm64 and x64 dmg files for latest version of Mayo(currently 0.9.0)
This tool would be really useful for dmg packaging: https://doc.qt.io/qt-6/macos-deployment.html
I don't have Apple hardware and have a virtual x64 macOS Monterey running in VirtualBox.
Mayo builds fine on that virtual macOS machine but opening document crashes for some unknown reason. I think it's related to bad OpenGL or graphics hardware emulation.
What would be great is having some people like you and @andrsd having Apple hardware to run and check the dmg files created.
Let me know if you agree to help testing this, if other people reading this message are interested then please drop a message in this thread.
I can provide a specific ci-macos.yml file to produce dmg file artifacts so someone can test, just at least one person willing to test

andrsd, 2025-01-20:
Just the output of the CI step is not enough. As @HuguesDelorme said, there needs to be one more step which is the deployment one. I do this for x86 version in one of my projects and it seems to work fine, see the relevant step

https://github.com/andrsd/mesh-inspector/blob/d8826143725f5541132550d765fc4b8d267fdc4c/.github/workflows/release.yml#L78-L83

Now that github has the arm64 runner images, this should be also possible for arm64, but I did not try yet.

However, keep in mind that the produced app bundles won't be signed, so users will see the that the app might be malware and will need to jump through several hoops to get the app going. This is fine for experiences mac users, but it will be a FAQ. 

If you wanted to sign the app, you need to participate in Apple Dev program (which is ~ $100 year), this will give you the signing certificates and you will need to bundle entitlements. With all that, the app will run smoothly. I do this locally on my machine when deploying my project for arm64. IDK if this can be automated via github action - I did not bother to try.

HuguesDelorme, 2025-01-23:
Hello @wormyrocks and @andrsd 
I've added some steps to macOS CI to build a dmg file for Mayo, can you download this artifact and let me know if it works?
https://github.com/fougue/mayo/actions/runs/12885554814/artifacts/2460863815

Or grab it from "Upload dmg file" step:
https://github.com/fougue/mayo/actions/runs/12885554814/job/35924263781

Many thanks!

andrsd, 2025-01-23:
I downloaded the artifact, installed from the dmg, ran it and got:

[image]

If I run from terminal, the binary is killed immediately, and in the report to Apple, I see:
```
System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x0000000102ff8000
Exception Codes:       0x0000000000000032, 0x0000000102ff8000

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page
```

andrsd, 2025-01-23:
That didn't change anything.

wormyrocks, 2025-01-27:
Try right clicking on the app and hitting 'open', that might give you the option to allow unsigned apps in system settings.

> Hello [@wormyrocks](https://github.com/wormyrocks) and [@andrsd](https://github.com/andrsd) I've added some steps to macOS CI to build a dmg file for Mayo, can you download this artifact and let me [...]

The two links in this post seem dead?

andrsd, 2025-01-28:
> Try right clicking on the app and hitting 'open',

This was what I did as a first step and that brought the message I posted above. 

> ...Mayo (built locally) runs fine on my M1 Pro MBP.

For me as well. In my experience, locally-built binaries are fine. You use your local toolchain to build the app, so the OS trusts it. The problem is when you move that binary to another machine. At least that's what I see on my end.

I vaguely remember seeing some app (it was either inkscape or gimp) that reported as being unsigned (or from untrusted developer, I don't remember exactly), but I could still install it. IDK how they deploy for macOS, thou...
Everything that I found for Qt ended up with a step where one had to sign with Apple certificate. And if one wanted no error/warning messages - notarization was also needed...

IIRC, the requirements from Apple on signing, entitlements, and notarization were more and more strict with every release of macOS. I found a lot of outdated info on the topic when I was trying to deploy my app. Maybe with older versions of Xcode it is possible to work around the problem, but I was after a native arm64 build which is possible only with fairly recent Xcode versions.

Just my personal experience with the subject. YMMV as usual...