fougue/mayo #317: IGES import freeze on incmesh

Mannshoch, 2025-01-14:
I try to import a 900MB IGES. The import stop after running incmesh a while (Progress 52%). 
There is no error. The only indication that something happens is because CPU usage goes down to 0%
Click on the red button to abord does not work. Mayo has to be force closed.

Tested in Mayo 0.8 and 0.9

HuguesDelorme, 2025-01-20:
Hello @Mannshoch 
It's hard to investigate on that problem without the input file
Can you provide a link to the IGES file?

Mannshoch, 2025-01-20:
Sorry, im not allowed to. But If you tell me how I could create a log I will do it.

HuguesDelorme, 2025-04-03:
@Mannshoch 
Could this be the same issue as https://github.com/fougue/mayo/issues/255?
#255 was fixed in current development version(coming v0.10.0) in the beginning of february.
If you can build Mayo from develop branch, give a try to load your IGES file.
Otherwise let me know and I can provide you a fresh installer of latest dev version