fougue/mayo #327: May I ask how to obtain the `GIF demo` in the README document?

yf-1994, 2025-02-22:
Hello!

I really like this project. How can I obtain the `GIF demo` in the `README` document? I want to learn through it.

Thank you.

HuguesDelorme, 2025-02-24:
Hello thanks your words!
Not sure I understood what you want. If you are asking about how to create an animated gif: I used ScreenToGif tool on Windows
Does that help?

yf-1994, 2025-02-25:
Hello!
Thank you very much for your reply. I apologize for not being able to express my intention clearly. What I want to say is how to obtain a ready-made model like that in readme, because I am not very good at handling it myself, and I want to use a ready-made model to observe its various characteristics

---- Replied Message ----
| From | Hugues ***@***.***> |
| Date | 02/24/2025 16:39 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [fougue/mayo] May I ask how to obtain the `GIF demo` in the README document? (Issue #327) |

Hello thanks your words!
Not sure I understood what you want. If you are asking about how to create an animated gif: I used ScreenToGif tool on Windows
Does that help?

—

Hello thanks your words!
Not sure I understood what you want. If you are asking about how to create an animated gif: I used ScreenToGif tool on Windows
Does that help?

HuguesDelorme, 2025-02-25:
You can create an account on [grabcad](https://grabcad.com/library) and search for step files to download

yf-1994, 2025-02-25:
okay!thanks so much!

---- Replied Message ----
| From | Hugues ***@***.***> |
| Date | 02/25/2025 22:57 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [fougue/mayo] May I ask how to obtain the `GIF demo` in the README document? (Issue #327) |

You can create an account on grabcad and search for step files to download

—

You can create an account on grabcad and search for step files to download