fougue/mayo #328: Can add a Chinese translation?

chenpaner, 2025-02-26:
Thank you very much for this handy software!
I really want to add a Chinese translation to this software, but I only know a little Python and don't understand C++ at all. So could you please provide an English dictionary for me? I can translate it into Chinese, and then we can add a Chinese interface.

HuguesDelorme, 2025-02-26:
Hello @chenpaner 
Providing a chinese translation for Mayo would be great!
The translation files are in Qt format(`*.ts`) and stored in [i18n](https://github.com/fougue/mayo/tree/develop/i18n) folder. There are two languages provided so far: english(`i18n/mayo_en.ts`) and french(`i18n/mayo_fr.ts`)

I've created an empty translation file for chinese: [mayo_zh.zip](https://github.com/user-attachments/files/18982535/mayo_zh.zip)
You can edit this new `mayo_zh.ts` file with the Qt linguist tool, easily available here: https://github.com/thurask/Qt-Linguist/releases
Download [linguist_5.15.16.zip](https://github.com/thurask/Qt-Linguist/releases/download/20241116/linguist_5.15.16.zip) and execute the `linguist.exe` file within the ZIP archive.
Then from Qt linguist just open `mayo_zh.ts` and start translate the messages.
Here is the manual of this tool, but it's rather intuitive to use: https://doc.qt.io/qt-6/linguist-ui.html

Feel free to ask your questions here in this thread, if there's any issue!

Once you have translated a few messages into chinese, then contact me and we will then add your `mayo_zh.ts` file and do the necessary stuff to make Mayo aware of this new language file.
After that you will be able to complete `mayo_zh.ts` file and check on your side how it looks like in Mayo

chenpaner, 2025-02-27:
Hello @HuguesDelorme ,
This is the mayo_zh.ts file that I translated. I checked each one and the results are shown in the following picture. If there are any problems, please let me know!
[image]

HuguesDelorme, 2025-02-27:
Hello @chenpaner 
That's looking great! Can you attach file `mayo_zh.ts` to this thread?

chenpaner, 2025-02-27:
> Hello [@chenpaner](https://github.com/chenpaner) That's looking great! Can you attach file `mayo_zh.ts` to this thread?

I just re-edited the reply and added a download link.

HuguesDelorme, 2025-02-28:
Hello @chenpaner 
I've just pushed branch i18n/#328 which contains your changes and provides the chinese translations.
This is looking fine, many thanks for your great and hard work!

Here are a couple of screenshots:
[image]
[image]

HuguesDelorme, 2025-02-28:
Merged to `develop` branch

closed 2025-02-28 by HuguesDelorme

chenpaner, 2025-03-01:
@HuguesDelorme 
I'm very sorry, but I have two more questions to ask. However, I don't think it's necessary to start a new issue conversation, so I'll continue to reply here!

1. How long do we have to wait for the release of the develop file with Chinese translation? I mean, how long will it take to download Mayo-1.0.0-win64-binaries.zip? Because I recorded a video to recommend this software on Chinese video websites, so I'm planning to wait for the new version to be released to make it easier for Chinese users to use this software in my video.

2. Also, I created a plug-in for importing CAD models in Blender that relies on this software. Specifically, I used mayo-conv.exe to convert models in the background. But I'm wondering if any errors occur during the conversion process with mayo-conv.exe, will there definitely be a return statement with "Error" in the console, such as "Error during import of 'D:\a'a'aaaa.stp' File read problem"? Since I can't predict all errors, I can only use the return statement "Error" to indicate a conversion failure and stop the operation!
3. The first version of mayo_zh.ts that I provided may not be very accurate. Maybe I need to modify some entries in the future. Should I reply to you with the new file directly here?

HuguesDelorme, 2025-03-01:
Hello @chenpaner

1. There's a single release of Mayo each year. Latest version is v0.9.0 which was shipped this year on january the 13rd. It takes time to develop a sufficient amount of noticeable features. Before reading your message I would have said that next version(v0.10.0) will we be released not before september or october this year. What's possible is to release a patch version(v0.9.1) early in the next couple of weeks, this version would contain only bufixes and the chinese translation you made
2. Didn't verify but in case of errors `mayo-conv.exe` should exit with code != 0. That said I've planned to rework mayo-conv and its output text, add some output that would be easily parsable by consumer applications
3. Yes for next improvements and fixes on `mayo_zh.ts` please provide pull requests through the GitHub interface. That will be more efficient and moreover you will appear in the list of contributors