fougue/mayo #330: XIO:  fatal IO error 62 (Timer expired) on X server ":0"

hramrach, 2025-03-14:
When opening multiple files I get:

WARNING: The X11 connection broke: I/O error (code 1)
XIO:  fatal IO error 62 (Timer expired) on X server ":0"
      after 237268 requests (237260 known processed) with 0 events remaining.
*** ERR StepReaderData : Unresolved Reference : Fails Count : 1 ***
Segmentation fault (core dumped)

hramrach, 2025-03-14:
The problem is likely opening many files with trivial geometry.

When a complex file is read there is some progress indicator displayed and that resets the timer.

However, when many files that are not complex enough to display import progress are read the error above is shown.

hramrach, 2025-03-14:
No, it's just the number of files.

Even with directory of files that would sometimes show progress when all files are read no progress is shown and mayo crashes.

Some progress for opening files is needed or lifting the requirement that the display needs to be always updated.

HuguesDelorme, 2025-03-14:
Hello @hramrach
Can you describe the exact steps to reproduce the crash? That input would be very valuable
Please, if possible, provide the files you used also. If not then describe the contents of these files

hramrach, 2025-03-14:
Use the open function, select all files in a directory -> crash

hramrach, 2025-03-14:
[image]

hramrach, 2025-03-14:
Cannot upload the files, unfortunately.

HuguesDelorme, 2025-03-14:
How many files did you load? All of STEP format?
What is the rough size of each?

hramrach, 2025-03-14:
-rw-r--r-- 1 hramrach users  105K Mar 13 09:52 Button_Switch_THT.3dshapes/DIP-10_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  122K Mar 13 10:14 Button_Switch_THT.3dshapes/DIP-10_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  128K Mar 13 09:53 Button_Switch_THT.3dshapes/DIP-12_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  147K Mar 13 10:14 Button_Switch_THT.3dshapes/DIP-12_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  143K Mar 13 09:54 Button_Switch_THT.3dshapes/DIP-14_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  166K Mar 13 10:15 Button_Switch_THT.3dshapes/DIP-14_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  158K Mar 13 09:55 Button_Switch_THT.3dshapes/DIP-16_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  184K Mar 13 10:15 Button_Switch_THT.3dshapes/DIP-16_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  173K Mar 13 09:56 Button_Switch_THT.3dshapes/DIP-18_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  203K Mar 13 10:15 Button_Switch_THT.3dshapes/DIP-18_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  189K Mar 13 09:57 Button_Switch_THT.3dshapes/DIP-20_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  221K Mar 13 10:16 Button_Switch_THT.3dshapes/DIP-20_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  204K Mar 13 09:59 Button_Switch_THT.3dshapes/DIP-22_W10.16mm_Socket.step
-rw-r--r-- 1 hramrach users  240K Mar 13 10:17 Button_Switch_THT.3dshapes/DIP-22_W11.43mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  205K Mar 13 09:58 Button_Switch_THT.3dshapes/DIP-22_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  240K Mar 13 10:16 Button_Switch_THT.3dshapes/DIP-22_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  220K Mar 13 10:01 Button_Switch_THT.3dshapes/DIP-24_W10.16mm_Socket.step
-rw-r--r-- 1 hramrach users  258K Mar 13 10:18 Button_Switch_THT.3dshapes/DIP-24_W11.43mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  220K Mar 13 10:02 Button_Switch_THT.3dshapes/DIP-24_W15.24mm_Socket.step
-rw-r--r-- 1 hramrach users  258K Mar 13 10:19 Button_Switch_THT.3dshapes/DIP-24_W16.51mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  220K Mar 13 10:00 Button_Switch_THT.3dshapes/DIP-24_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  258K Mar 13 10:18 Button_Switch_THT.3dshapes/DIP-24_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  250K Mar 13 10:05 Button_Switch_THT.3dshapes/DIP-28_W15.24mm_Socket.step
-rw-r--r-- 1 hramrach users  294K Mar 13 10:21 Button_Switch_THT.3dshapes/DIP-28_W16.51mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  251K Mar 13 10:03 Button_Switch_THT.3dshapes/DIP-28_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  295K Mar 13 10:20 Button_Switch_THT.3dshapes/DIP-28_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users   40K Mar 13 10:12 Button_Switch_THT.3dshapes/DIP-2_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  289K Mar 13 10:07 Button_Switch_THT.3dshapes/DIP-32_W15.24mm_Socket.step
-rw-r--r-- 1 hramrach users  339K Mar 13 10:22 Button_Switch_THT.3dshapes/DIP-32_W16.51mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  289K Mar 13 10:06 Button_Switch_THT.3dshapes/DIP-32_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  339K Mar 13 10:21 Button_Switch_THT.3dshapes/DIP-32_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  350K Mar 13 10:08 Button_Switch_THT.3dshapes/DIP-40_W15.24mm_Socket.step
-rw-r--r-- 1 hramrach users  414K Mar 13 10:23 Button_Switch_THT.3dshapes/DIP-40_W16.51mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  412K Mar 13 10:09 Button_Switch_THT.3dshapes/DIP-48_W15.24mm_Socket.step
-rw-r--r-- 1 hramrach users  492K Mar 13 10:24 Button_Switch_THT.3dshapes/DIP-48_W16.51mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users   50K Mar 13 09:50 Button_Switch_THT.3dshapes/DIP-4_W10.16mm_Socket.step
-rw-r--r-- 1 hramrach users   57K Mar 13 10:13 Button_Switch_THT.3dshapes/DIP-4_W11.43mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users   49K Mar 13 09:50 Button_Switch_THT.3dshapes/DIP-4_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users   56K Mar 13 10:12 Button_Switch_THT.3dshapes/DIP-4_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  543K Mar 13 10:11 Button_Switch_THT.3dshapes/DIP-64_W15.24mm_Socket.step
-rw-r--r-- 1 hramrach users  649K Mar 13 10:25 Button_Switch_THT.3dshapes/DIP-64_W16.51mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users   74K Mar 13 09:51 Button_Switch_THT.3dshapes/DIP-6_W10.16mm_Socket.step
-rw-r--r-- 1 hramrach users   85K Mar 13 10:13 Button_Switch_THT.3dshapes/DIP-6_W11.43mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users   74K Mar 13 09:50 Button_Switch_THT.3dshapes/DIP-6_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users   85K Mar 13 10:13 Button_Switch_THT.3dshapes/DIP-6_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users   90K Mar 13 09:52 Button_Switch_THT.3dshapes/DIP-8_W10.16mm_Socket.step
-rw-r--r-- 1 hramrach users  103K Mar 13 10:14 Button_Switch_THT.3dshapes/DIP-8_W11.43mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users   89K Mar 13 09:51 Button_Switch_THT.3dshapes/DIP-8_W7.62mm_Socket.step
-rw-r--r-- 1 hramrach users  103K Mar 13 10:13 Button_Switch_THT.3dshapes/DIP-8_W8.89mm_SMDSocket_LongPads.step
-rw-r--r-- 1 hramrach users  192K Mar 13 09:48 Button_Switch_THT.3dshapes/Nidec_Copal_SH-7010C.step
-rw-r--r-- 1 hramrach users   39K Mar 13 10:12 Button_Switch_THT.3dshapes/SW_DIP_SPSTx01_Piano_10.8x4.1mm_W7.62mm_P2.54mm.step
[... cut ...]
-rw-r--r-- 1 hramrach users  104K Mar 13 09:48 Button_Switch_THT.3dshapes/SW_PUSH_6mm_H8.5mm.step
-rw-r--r-- 1 hramrach users  152K Mar 13 09:48 Button_Switch_THT.3dshapes/SW_Tactile_Straight_KSA0Axx1LFTR.step
-rw-r--r-- 1 hramrach users  152K Mar 13 09:48 Button_Switch_THT.3dshapes/SW_Tactile_Straight_KSL0Axx1LFTR.step
-rw-r--r-- 1 hramrach users   95K Mar 13 09:48 Button_Switch_THT.3dshapes/SW_TH_Tactile_Omron_B3F-10xx.step
-rw-r--r-- 1 hramrach users  104K Mar 13 09:48 Button_Switch_THT.3dshapes/SW_TH_Tactile_Omron_B3F-11xx.step

hramrach, 2025-03-14:
Looks like github has some strict limit on attachment size, cannot add 7.7MB file.