fougue/mayo #332: Different STEP File Structure in Mayo-XDE and DFBrowser

dabari, 2025-03-22:
@HuguesDelorme thank you for your great work and this app.

**Description:**
When I open this simple STEP file in Mayo (see below) using the XDE inspection, I see shape structures that don’t make sense to me – they seem to be organized differently than expected. For comparison, I opened the same STEP file in the OpenCascade DFBrowser. There, the structure looks correct and as expected.

[image]

**Expected Behavior:**
The STEP file in Mayo should be displayed in the same way as in DFBrowser, with the shape structure matching the original model.

**Steps to Reproduce:**
* Load the STEP file in Mayo (tested with version 0.9.0)
* Start XDE inspection
* Check the shape structure
* Open the same file in DFBrowser and compare

**Question / Request:**
Can you check why the STEP structure looks different in Mayo-XDE compared to DFBrowser?

Thanks for your answer.

[test.stp.txt](https://github.com/user-attachments/files/19403254/test.stp.txt) -> rename to stp

dabari, 2025-03-23:
I’ve observed that using the hot reload feature (#153) leads to this strange behavior in the XDE inspector.

When a file is modified in the background and Mayo reloads it via hot reload, the XDE inspector displays partially incorrect or inconsistent entries.

I suspect that the internal model or `TDocStd_Document` isn't being updated properly – old elements may not be fully removed, or new ones may not be correctly integrated. As a result, the XDE inspector might end up showing a mixed or corrupted state.

Steps to Reproduce:
* Open a STEP file in Mayo (the first time) → the XDE inspector shows the correct structure.
* Modify the currently opened STEP file in the background to trigger the hot reload function.
* Reopen the XDE inspector → the displayed structure is inconsistent or incorrect.

Jopie01, 2025-03-27:
Hmm, I wanted to report a weird thing and came across this. I also think this causes also some weird behavior when you have clipped the model (section) or do some measurements. Then make some changes to the 3D model and export it again. After reload you have to close the measurement or select another measurement type. Same for the section and also when you switch on or off parts in an assembly. You have to make a change and then everything works again.

closed 2025-04-03 by HuguesDelorme