fougue/mayo #347: Image\cameraOrientation output

merco, 2025-05-07:
I was wondering (as a newbie) how to get a view oriented in this way

[1M001934.zip](https://github.com/user-attachments/files/20077520/1M001934.zip)

[image]

the default in mayo seems 180° Z-Axis rotated....

[image]

HuguesDelorme, 2025-05-07:
Use this orientation: `X -1   Y 1   Z 1`
There's currently no parameter for the display mode with the Image exporter...

HuguesDelorme, 2025-05-07:
Yes adding a `displayMode` parameter to the Image exporter is possible, rather easy I would say

merco, 2025-05-07:
Another question about image output  / display .
This glb file is converted from step. "mld2050263.png" was taken during the step conversion, the "aa.png" from the glb file.

I notice some differences in rendering of materials/colors and then a black part that actually should be colored

[image]

[image]

[mld2050263.zip](https://github.com/user-attachments/files/20085371/mld2050263.zip)

merco, 2025-05-12:
Hi, @HuguesDelorme any ideas about the material color issue?

HuguesDelorme, 2025-05-12:
Actually the part you mentionned in the GLB file is not black but cyan as expected, try to zoom in:
[image]

There's an overlaying frame on it, don't know from where it came from during the convesion

merco, 2025-05-12:
@HuguesDelorme maybe is a "shaded with edge" related issue? Can it be set with settings.ini ?

HuguesDelorme, 2025-05-12:
Yes switching to "Shaded" display mode is better in such case:
[image]

merco, 2025-05-13:
@HuguesDelorme ok, is it possible to have that setting (Display Mode) availabe in the configuration file?

closed 2025-05-16 by HuguesDelorme