fougue/mayo #354: Importing Large .stp Files causes to stop Mayo

OleskyBogdanov, 2025-08-06:
I am trying to import a large .stp file (~2.4 GB) into Mayo. After waiting for around 4 hours, the application just stopped working without completing the import. This happens on both macOS 15.6 and Ubuntu 22.04.5 LTS.

Unfortunately, I cannot share the .stp file due to confidentiality.

Are there any settings, tweaks, or recommendations to improve performance when dealing with very large .stp files?

HuguesDelorme, 2025-12-30:
Do other softwares can open your .stp file? Freecad or CAD Assistant for example(they also rely on OpenCascade)

OleskyBogdanov, 2026-03-20:
I'm sorry for the long wait for an answer. Tested on less large file (700mb), FreeCad crashed, Mayo and CAD Assistant opens file. On ~2.4 GB FreeCad and CAD Assistant crashed