fougue/mayo #359: run error

271812697, 2025-08-30:
when i build mayo by cmake. i have finish all steps! and run it! 
i always meet the following errors!

[image]

I have tried several times, and although the program compiles successfully, I always encounter a missing dll error when running it. I can't wait to read the source code of Mayo over and over again, but when the program doesn't run, and make it  difficult to do such a thing.

HuguesDelorme, 2025-09-01:
Hello Buranya, if you build with VisualStudio then in cmake-gui, check option `Mayo_PostBuildCopyRuntimeDLLs` is ON.
Looks like tbb12.dll is not accessible from PATH

If you're blocked then use pre-built binaries or installer. See [Releases](https://github.com/fougue/mayo/releases) page

271812697, 2025-09-02:
This issue does seem to be related to OCC. The OCC I downloaded from the link in the build guide does not have these dlls. And the project built with VS2022 does not have the OCC target! Manually copying the dll can indeed solve this problem!

closed 2026-03-17 by HuguesDelorme